oresat/oresat-structure #1: Identifying parts from other repos

Joedang, 2017-03-14:
The README mentions that parts from the reaction wheel repo are used. If we don't need to import those from the STEP files, that would probably save a lot of time. I'm going to guess the whole reaction wheel assembly is used?

JSRoland, 2017-03-14:
Yes. I agree.

jalouke, 2017-03-14:
The files that are referenced from the structures repo aren't *.step they are *.sldprt and *.sldasm. The dependencies in the top level cubeSAT structure that I created only require that both repos be in the same directory. If that is not the case, then please let me know and I will fix it.

jalouke, 2017-03-14:
I just saw what you meant by step files, and I want to say...

Oh god no, I am not actively a part of OreSAT anymore, but a solid tip about project management, everyone should use the same software when working on a solid model. It is hard enough to make sure there are no conflicts when multiple people are working on the same assembly. It makes everyone's jobs that much harder having to convert files back and forth between different file types, and then praying that there aren't any conversion issues (when converting STEP files to sldprt files, sometimes the part loses surfaces and has to be repaired before it can be used).

Everyone should be using solidworks, or everyone should be using something else. If there is a legitimate concern that Solidworks doesn't have a feature, or there is an easier piece of software to use then by all means switch. However everyone needs to be on the same page.

Joedang, 2017-03-14:
>  everyone should use the same software when working on a solid model.
> Everyone should be using solidworks

Yep.

@calvin-young Did a big rework of the satellite structure in some other program and we're working on porting it into SW so we can all work on it.

jalouke, 2017-03-14:
OK, I think I understand the original question better now. The answer is, it should ultimately be re-made, because I really only drew the version that was originally in the 2U model as a first draft. It also used the original board design as a mounting base (Which if I am not mistaken has changed). The only critical dimensions within the reaction wheel system are the angles of the reaction wheels (check the reaction wheels repo for the correct angles).

Joedang, 2017-03-14:
>  it should ultimately be re-made, because I really only drew the version that was originally in the 2U model as a first draft.

Okay, rad. Thanks.

closed 2017-03-14 by Joedang