oresat/oresat-structure #2: Migrate from STEP to SolidWorks

Joedang, 2017-03-14:
# SW migration
So, @calvin-young did an awesome refactor of the OreSat structure. Unfortunately, he did it in a different CAD program than most of us. Our task is to migrate that whole assembly into the SolidWorks format. 

## How to migrate a part
- If you open [`Rev v0.5/2U_CUBESAT_V0_5.stp`][Rev0.5] in SolidWorks, it will open as an assembly. 
  - SW will complain about invalid templates and ask you if you want to use a blank template. Select `Cancel`.
  - Select a template to use. The SW assembly tutorial template or the PSAS assembly template should work fine. 
- Go look at the [TODO in the main README](https://github.com/oresat/oresat-structure/blob/master/README.md) and find the first unchecked file. 
  - Confirm that the file you're working on hasn't already been migrated. If it has, update the TODO!
- In SolidWorks, find that part in the assembly -- the structure of the TODO matches the structure of the assembly. Right click on it and open the part by itself. 
- SW will ask you if you want to automatically identify the features of the part. 
  - Just for shiggles, do. It probably won't work but you might get lucky. 
    - The goal is for the first feature in the part to _**not**_ be called something like `Imported3`.
    - You can try recognizing only certain features types that you consider to be the 'last' features and just recursively do that on the 'Imported' feature until you are done. That sometimes works. 
  - If that doesn't work (and it probably won't), you'll have to identify the features interactively. **I still haven't figured out how to do this effectively. Maybe someone can write a short guide to it here.** Hopefully you won't have to reimplement the part from scratch.
- Now that you have the part implemented, save it with the _**same name**_ as the corresponding STEP file and check it off in the [README](https://github.com/oresat/oresat-structure/blob/master/README.md). It's important that you save it with the _**same name**_ and check it off in the [README](https://github.com/oresat/oresat-structure/blob/master/README.md) so it's easy to tell what's been migrated! It should be self-explanatory where to put your SW file in `v0.5_migrated/`. If you aren't sure, check the READMEs for `v0.5_migrated/` and its subfolders.

## Rebuilding the subassemblies
- Check that _**all**_ of the parts required for you subassembly have been ported to SW.
  - If they haven't, port them! (see above)
  - If they have, feel free to use `git mv` to rename them (Use underscores instead of spaces! Some commands don't do well with spaces.) to something more descriptive, if you want. Make a note in the TODO indicating what they've been renamed to. (If you just rename them rather than using `git mv`, git will think you deleted the file and made another one. Since git operates on the _differences_ in files, this can waste space if your file is large. It would also make it harder to follow the history of the file, if anyone ever needs to.)
- As above, open up [`Rev v0.5/2U_CUBESAT_V0_5.stp`][Rev0.5]. 
- Right click on the top-level assembly and hide it.
- Unhide the assembly you're building. 
- Open a new assembly.
  - Import the corresponding parts which have been ported to the SW format. (I recommend importing the biggest part first, since the first part you import doesn't move.)
  - Proceed to mate the parts together. Preferably, you should use mates which correspond to how the parts would mate in meatspace. 
  - Use the STEP assembly as a reference for where stuff goes. 
  - If you can still wiggle parts around with your cursor, you probably aren't done mating. ( ͡° ͜ʖ ͡°)
- When your assembly is done, save it with the same name as the corresponding assembly in the STEP file. (This is less of an issue with the assemblies, since the original assembly names and any other names you choose should be obvious.)
- Cross out the subassembly on the TODO. 
  - Use double tildes to indicate cross-out: `~~hello world~~` becomes ~~hello world~~.

[Rev0.5]: https://github.com/oresat/oresat-structure/blob/master/Rev%20v0.5/2U_CUBESAT_V0_5.stp "complete STEP file assembly of OreSat version 0.5"

Joedang, 2017-04-29:
Parts people are converting:
* HELICAL CAN
  * Josh
* SPACE FOAM
  * Yohannes
* CARD LOWER
  * Marie
* CARD CAMERA
  * Marie
* ANTENNA SOLARPANEL UNWRAPPED
  * Chris
* coordinate page 
  * Marie

jalouke, 2017-04-29:
I totally forgot about the meeting, what time was it?

Joedang, 2017-04-29:
Hey @andrewgreenberg, don't forget to add @hmarie2, @RocketmanYG, and whatever Josh Carlson's username is.

andrewgreenberg, 2017-04-30:
All done. I need Josh Carlson's github user's name, there are 5 josh carlsons on github.

Joedang, 2017-05-07:
Hey, @andrewgreenberg, you still need to add @hmarie2.

andrewgreenberg, 2017-05-08:
OMG I AM A TERRIBLE PERSON. Done.

Joedang, 2017-05-13:
Okay, if you were in charge of converting parts in a subassembly, you are now in charge of putting together that subassembly and putting it into the main assembly.

So to be clear, these people are making these subassamblies:
* HELICAL CAN
  * Josh (@paperman5)
* CARD LOWER
  * Marie (@hmarie2)
* CARD CAMERA
  * Marie (@hmarie2)
* ANTENNA SOLARPANEL UNWRAPPED
  * Chris (@wilsoncw6)

Joedang, 2017-05-20:
Mkay, Imma close dis.

Reason: The conversion is done and we've moved on to making changes again.

closed 2017-05-20 by Joedang