oresat/oresat-structure #3: Part Identification

Joedang, 2017-03-29:
# The Problem
There are a few parts in the OreSat assembly that don't seem to have descriptive names. If someone can identify them, we could probably just find some pre-made SW model. The both have notches, so there's only one way to plug them in. I didn't see any matching connectors when I searched online.  I'm guessing @calvin-young, @andrewgreenberg, or @wrh2 will recognize these. This is of course related to #2.

## 40 pin 
This is the 40-pin connector used on most of the boards.
### Male
name in assembly: 100211647UGM000_B.1
[image]
### Female
name in assembly: 1521040X401XXX_100632479MOD0_1
[image]

## 20 pin 
This connector is used on the solar panels and the backplane.
### Female
name in assembly: 1522020X601XXX_100210477MOD000B 
[image]
### Male
name in assembly: 100211587UGM000_B.2
[image]

andrewgreenberg, 2017-03-29:
These are two forms of the oreSat card to backplane connectors on OreSat. The 40 pin is what we started with, but quickly moved to the 20 pins. We're currently still on the 20 pins. Males go on the cards, females go on the backplane.

There's no formal name, but you could call them 20pin-oresat-card, 20pin-oresat-backplane, 40pin-oresat-card, and 40pin-oresat-backplane, for example. Those long alphanumeric codes are their generic part names.

Hope that helps?

andrewgreenberg, 2017-03-29:
Note that they came from the manufacturer as STEP files, they were never SW directly.

Joedang, 2017-03-29:
Do you know who makes them? Searching for those part numbers doesn't turn up anything.

andrewgreenberg, 2017-03-29:
They're Harting "har-flex" (seriously?) board to board & mezzanine connectors.

- Harting "Board to Board & Mezzanine Connectors har-flex THR STRAIGHT FEMALE **6.25mm** 40pin PL1", manufacturer pert number (MPN) 15210402401000
   - http://www.mouser.com/ProductDetail/HARTING/15210402401000/?qs=sGAEpiMZZMvlX3nhDDO4AKHv1woNlxZg7ELnbOpcf0U%3d
   - https://b2b.harting.com/ebusiness/app/displayApp/(layout=7.0-12_132_124_140_130&carea=0001400888&citem=00014008880001414518)/.do?rf=y

- Harting "Board to Board & Mezzanine Connectors har-flex RA MALE 40pin PL1" 
   - https://b2b.harting.com/ebusiness/app/displayApp/(layout=7.0-12_132_124_140_130&carea=0001397574&citem=00013975740001397570)/.do?rf=y
   - PL1/PL2 and Tape Reel doesn't really matter
   - http://www.digikey.com/product-search/en/connectors-interconnects/rectangular-connectors-headers-male-pins/1442547?k=1515040&k=&pkeyword=1515040&pv7=2&mnonly=0&newproducts=0&ColumnSort=0&page=1&stock=1&quantity=0&ptm=0&fid=0&pageSize=500

Joedang, 2017-03-29:
Thank you slack daddy.

Joedang, 2017-10-05:
So, I was putting together the tail card (the thing in position 19 that serves as a mezzanine for the -Z endcap). I grabbed [this model](https://b2b.harting.com/ebusiness/en_us/har-flex-ang-m-12p-PL1-560pcs/15150122601000) (ignore the picture, it really is a 12 pin connector) to go on the tail card and connect to the solar panels on the -Z unit.

It looks like the female connector on the solar panels (bright yellow) is intended to be used with a male connector (dark blue) that has a longer jacket. I couldn't seem to find a longer-jacketed connector on the Harting site though. I think it probably doesn't matter, but it might be a good idea to go find that longer connector. 

[image]

andrewgreenberg, 2019-04-29:
This should all be done by now!

closed 2019-04-29 by andrewgreenberg