oresat/oresat-structure #4: Connector Alignment in Step Assembly

Joedang, 2017-03-29:
#  The Problem
It appears that the connections between the generic board and the backplane are slightly misaligned. The backplane connectors are a little high (0.04 mm to 0.1 mm). I'm not sure how big of a deal that is, maybe the boards are elastic enough to accommodate it; maybe there's something weird about the way the parts were mated; maybe there's going to be more error from the manufacturing. 

(obviously related to #2 )

## MMCX misalignment (0.04 mm)
You can also see that the inner conductor of the female connector doesn't have a hole in it, but it appears it should (right center). I'm also not sure if the male connector is supposed to have some barb on it to attach to the barb on the female connector.  
The red stuff is interference.
[image]

## 40 pin misalignment (0.1 mm)
The horizontal alignment is fine (top left), but the vertical alignment is off (right). There's also no hole modeled in the female connector (bottom left). That doesn't really matter, but it could be annoying later on when we check for interference in the assembly. 
[image]

# What to do
Maybe someone like @glennl or @andrewgreenberg will know off the top of their heads whether 0.1 mm is negligible for connector alignment. 

We might have to tweak the design of the backplane, which doesn't seem like too big of a deal since it's not in active development right now. I was going to mate the generic board with the rack+backplane so I could figure out which parts of the board are off-limits for components. I was worried that stuff might change, but it looks like this is resolvable by only making changes to the backplane.

andrewgreenberg, 2017-03-30:
Ish. We care about things that are > 0.1 mm off. But 0.04 mm? Probably not. 

That said, there's no reason to not exactly align them. The backplane  hasn't been fabbed yet, so just go ahead and set the correct height for the connectors on the  backplane. When everything is done, I'll take a drawing and update the eCAD to whatever you specify.

glennl, 2017-03-30:
I think >= 0.1 mm does matter as Andrew said, particularly when thermal
changes and vibration issues come in to play.  It is possible the
connectors ordered are the problem. Often connectors of the same series can
have somewhat different dimensions, ever so slightly. My dream-state tells
me, it would be nice if critical dimensions could be adjusted slightly
during board assembly. ;)

- Glenn

jalouke, 2017-03-30:
I think that the through holes for those plugs have more than .1mm slop
(the thickness of a human hair) in them. However it still should still be
fixed, because there is no reason to have "slop" in CAD.

> Ish. We care about things that are > 0.1 mm off. But 0.04 mm? Probably
not.
>
> That said, there's no reason to not exactly align them. The backplane
> hasn't been fabbed yet, so just go ahead and set the correct height for
the
> connectors on the backplane. When everything is done, I'll take a drawing
> and update the eCAD to whatever you specify.

> <https://github.com/oresat/oresat-structure/issues/4#
issuecomment-290263078>,
G4J_dOkkRks5rqvIOgaJpZM4MtszW>

Joedang, 2017-03-31:
> it still should still be fixed, because there is no reason to have "slop" in CAD.

Yeah, I'll add a note to the backplane in the TODO. It seems like it should be pretty straight forward to fix.

Joedang, 2017-05-21:
So, tiny update to this. It was unclear _which_ model/manufacturer of MMCX connectors were used. AFAIK, @jalouke made the models for the connectors based on a manufacturers diagrams. Since I wasn't sure of the model/manufacturer, I'm saying these ([male](https://www.digikey.com/product-detail/en/samtec-inc/MMCX-P-P-H-ST-TH1/SAM10618-ND/6561612), [female](https://www.digikey.com/product-detail/en/samtec-inc/MMCX-J-P-H-RA-TH1/SAM10615-ND/6561608), [TDS](http://suddendocs.samtec.com/catalog_english/mmcx.pdf), [male CAD](https://www.samtec.com/partnumber/mmcx-p-p-h-st-th1?vendor=digikey), [female CAD](https://www.samtec.com/partnumber/mmcx-j-p-h-ra-th1?vendor=digikey)) are canon. They are also Greenberg approved.

jalouke, 2017-05-22:
they should be named after their respective part numbers on digikey.

Joedang, 2017-06-04:
I looked around some of the older commits. It seems the MMCX connectors either have a generic name or aren't used in the assembly. If there's no strong reason to seek out what's in the model, I'm just going to replace them with the Samtec connectors.

andrewgreenberg, 2017-06-04:
Actually, not the digikey part number, but the <manufacturer>-<manufacturer's part number>.

Also the Samtec connectors *are* MMCX connectors... so maybe we should name the parts mmcx-samtec-<partnumber>.

Joedang, 2017-06-04:
The file names that Samtec gave the connectors are `MMCX_P_P_H_ST_TH1.SLDPRT` and `MMCX_J_P_H_RA_TH1.SLDPRT`, so I'm just gonna leave them like that. 

Reason for closing: misalignments have been fixed.

closed 2017-06-04 by Joedang