oresat/oresat-structure #5: Clean up Directory Structure

Joedang, 2017-05-10:
## The Problem
Having all the design versions in different directories *while already being part of the repo* is silly. This is what Git is for. 

## Proposed Solution
- [ ] Wait until the structure is completely migrated into SolidWorks.
- [ ] For every revision `n`:
  - [ ] Go through the commit history and find the *first commit* where the `Rev v0.[n]` directory  appears.
  - [ ] Tag the commit *before that* as release `0.[n-1]`.
- [ ] Remove all the old revisions.
- [ ] Move the contents of `v0.5_migrated` to the top-level directory.
  - [ ] Remember to open the assemblies and re-link any parts which SW lost track of.
- [ ] Tag as release `0.5.1`.
- [ ] Close this issue.

## Remarks
The reason for waiting to change the directory structure until after the conversion is convenience. I could do it on another branch, but I'm not sure how Git would handle binary files which were changed on one branch and moved on another. It's also convenient to have the old and new models available at the same time during the conversion.

andrewgreenberg, 2017-05-11:
This would be amazing. Go ahead and tag it as something higher, like 0.6 or even the dreaded 1.0. We're done playing around, I think we're close enough that we really have something here.

Joedang, 2017-05-12:
I was thinking the criterion for `1.0` would either be "most/any changes would force other projects to change their work" or "this is something that might actually fly". The main things I would not call "1.0" are the turnstile antenna and the camera mounting.

Joedang, 2017-05-18:
The assemblies still need to be updated to point to the new file locations, but that should just happen automatically as people open them.

closed 2017-05-18 by Joedang