oresat/oresat-structure #8: Helical Can Connector Locations

paperman5, 2017-05-18:
The helical can subassembly in the old step assembly of the satellite has multiple connectors phasing through geometry without the appropriate cutouts.
[image]
I am not sure if all of these connectors are supposed to be where they are currently placed. Could someone with some electrical knowledge of the satellite give some input? For the new assembly I will only add the connector on the bottom of the plate for the time being.

paperman5, 2017-05-18:
@Joedang or others, would you mind referring some people who might know about this issue?

andrewgreenberg, 2017-05-18:
Me! Talk to me!

I assume those are connectors for the solar panels. So yes, they're still there. However, is this an end cap, or an actual card? Can you come Friday 2pm - 5pm to discuss?

Joedang, 2017-05-19:
@paperman5 Yeah, those are for the solar panels. If you take a gander at the v0.5 assembly, they match up with the connectors coming out of the plane of the solar panel boards. 

@andrewgreenberg I think this is showing the 2nd-to-last card on the +Z end. I assume we don't want the helical antenna to be connected to its ground plane though. I guess we have details to work out on both antennas.

paperman5, 2017-05-19:
@andrewgreenberg @Joedang I won't be able to make it to the Friday meeting, although I might be able to jump on Hangouts for a while. This is the last card that I see on the end of the old step assembly, though there's probably a cap that goes over it. If the connectors need to be there I can include some cutouts in the part, but I imagine that would make the card harder to manufacture.

That's the way the antenna was attached in the old assembly, I can change it once we get some feedback.

paperman5, 2017-05-31:
Closing this since the antenna is being redesigned so this isn't relevant anymore

closed 2017-05-31 by paperman5