oresat/oresat-structure #10: Helical Endcap Fastening

paperman5, 2017-05-31:
There is currently no way that I see for the helical antenna end cap to be fastened to the end of the satellite:
[image]
A possible way to fix this would be to add supports underneath the endcap coming out of the sides of the rail assembly. However I need to confirm that this is allowable as I am not sure of the rail requirements and it would impose some (still generous) vertical restrictions for the solar panel card that I believe goes under the endcap. I was thinking something like this:
[image]
Does this seem reasonable? If it does, I will go ahead and make these changes.

andrewgreenberg, 2017-06-01:
Huh, this is an interesting problem. First, the end cap has to be mostly a solid PCB, since it's the ground plane for the helical antenna. Second, the helical antenna can't be "embedded" in the PCB, it has to go start from that ground plane. So you'll have to drop the end cap "into" the satellite body if you need more than the 6.5mm height from the top of the frame to the top of the rails.

Your proposal seems fine to me. I approve if Joe approves. One thought: can we squish "out" the coiled up helical so it fits in the 6.5 mm space? That means slightly "unwinding" the helical as it gets compressed, but that seems OK?

Joedang, 2017-10-06:
So, I made some endcap boards that just sit on top of the structure. They screw into the rack and register with the rails with 0.1 mm of clearance (so, 0.2 mm of slop in X and Y).

It's worth noting that with the mezzanine connections to the head/tail cards (see #12 and #13), we loose some real estate for antennas.

## +Z face:
[image]

## -Z face:
[image]

## -X corners:
[image]

## +X corners:
[image]

Joedang, 2017-10-07:
Update from the Greenberg:  
Osh Park guarantees features that can be done with a 0.1" (yes, 1/10 inch, not 0.1 mm) endmill. So, the dogbones would need to be no larger than 0.05" (1.27 mm) in radius. I think they're 1 mm in radius right now, so I'll have to go in and change them.

Joedang, 2018-01-23:
I updated the radius of the dogbone fillets. The original issue has been addressed, and I assume that any if additional changes are needed, they will be brought up by the capstone. [So, here I go closing again. ](https://youtu.be/YG6DifUtPvs)

closed 2018-01-23 by Joedang