oresat/oresat-structure #11: Choose Inhibit Switches

Joedang, 2017-06-04:
There was some good discussion on how to do the inhibit switches in the hangout today, so I kind of want to capture what we worked out and what needs to be worked out in the form of an issue. 

# Problem
The Nanoracks spec requires us to have inhibit switches. We don't have any in our design yet. 

Inhibit switches are similar to the remove-before-flight pin. They keep the satellite from turning on. When the satellite is in the peapod, the switches are activated (presumably in a way that crowbars satellite power). When it gets launched, the switches pop out into the closed state, giving power to the satellite. After waiting 30 minutes, a simple board on the satellite tells everything to start turning on. 

## [Requirements as per the Nanoracks SRD-029 section 4.7 (page 12)][SRD-029]:
* We need at least 3 inhibit switches.
* They can be pushers in the -Z rail ends, or roller/sliders in the sides of the rails.
* Any roller/sliders must be at least 75% of the width of _the deployer's guide rail_, aka at least 1.875 mm (much smaller than we thought when discussing on the hangout).
* The switches cannot exert more than 3 N. (That's 305 grams-force, for anyone searching Digi-Key.)

It's worth noting that the spec doesn't say if that 3 N is the _total_ force of all the switches or for a single switch. Two of them will likely be on opposing faces though, so it's a moot point. 

# Initial thoughts/solutions
Everyone basically agreed that putting switches in the -Z ends of the rails would be awful. 

The thought for a while was that we could embed lever-style limit switches (like those typically seen in 3D printers or laser cutters) into the rails. However, this would be a kind of big change to the structure of the satellite and require weird changes to the card slots and a very custom board(s). 

We thought about using "cross roller plunger switches", but they're all kind of huge and would require big cutouts from the structure. "Simulated rollers" were also a nice option, but violate that 75% rule. 

The general consensus between @andrewgreenberg, @jalouke, @paperman5, and @joedang at the end of the hangout seemed to be that we should use a plunger or lever switch on one of the boards and make an extender for it. The extender would go through a hole in the rail. (see below)

# Mockups
I made a thing in SolidWorks to get an idea of how we might implement the inhibit switches in OreSat. 

I used [Panasonic's ESE-11SH1C][SH1C] and designed for a 1.5 mm travel. (The [ESE-11SH2C][SH2C] may also come in handy if we choose to use that switch.) The data sheet for the switch says it triggers at 0.75 mm and bottoms out at 2.4 mm. We can't really make the travel more than 2 mm, otherwise we'll either have a chance of the switch sticking when it goes to uninhibit mode or when it's inserted into the peapod. (see 2nd and 5th images for clarification)

Using this style of switch, we can add a spring if we want more force.

We would need to clip the leads of the switch after soldering it into the board, otherwise they would collide with the rack. That's NBD.

If we want to center the plunger on the plunger extender (I do), we'll have to make a cutout in the card rack to accommodate it. The extender is already centered on the rail. The extender _must_ be centered on the peapod guide rail. AFAIK, the guide rail is centered on the cubesat rail (Figure 9 in SRD-029 seems to agree. Someone please give me a sanity check here.)

## uninhibited state:

[image]

[image]

[image]

### inhibited state:

[image]

[image]

[image]

# Action to take
* Since @hmarie2 has been reading about switches, maybe she can determine whether or not this would be an acceptable switch to use and/or look for alternatives.
* If we do want to do something like this, we should think about where all the limit switches will go, since they will need corresponding cutouts in the rails/rack. 
* We should consider how we would go about manufacturing a plunger extender and making the cutouts in the rails/rack. (eeny-weeny engraving endmill and CNC, probably)
* How can we reduce the chance of the switch sticking?
* Is this switch non-weeny/low-resistance enough for the ECEs? (I'm assuming it's not really a high-power deal.)

[SRD-029]: https://drive.google.com/open?id=0B_LcfbAtlfiZQkhqN1R0VHQybFU
[SH1C]: https://www.digikey.com/short/3dbb78
[SH2C]: https://www.digikey.com/short/3dbb31

Joedang, 2017-06-04:
Also, side note, is this sort of thing more appropriate as an issue or a project?

jalouke, 2017-06-04:
I think we were mistaken when we were discussing the size of the portion in
contact with the P-pod. I believe it needs to be 75% of the with of the
P-pod rail, not the cubeSAT rail ( 75% of 2.5mm).

Joedang, 2017-10-06:
A little bit of an update to this issue:  
We picked out some reasonable switches. (They're a bit big, but smaller switches just have too small of travel.) Based on the manufacturer's data sheet, various travel distances of the switch are good but the tolerance is too loose for what we need. We ordered some of them and it looks like they reported very conservative tolerances. (In reality, they are much tighter.) I went through and measured the actuation positions with calipers (not sure if I pushed that data). It would be nice to redo that with a micrometer.

I still need to do the tolerance stack to get the ideal dimensions for the plunger, bracket, and mounting holes. (Maybe I should make a notebook for that? I did it as a SW drawing, but that turned out to be almost useless.)

[image]

andrewgreenberg, 2019-04-29:
This is now Oleg's! Not sure if the above discussion is useful, but just in case.

catiespivey, 2019-09-25:
Can we close this since the design is done? Just waiting on you and David to do testing @andrewgreenberg

closed 2020-07-12 by EmilioGizzi