oresat/oresat-structure #12: RF connections to +Z and -Z endcaps

Joedang, 2017-10-02:
# Problem
We need a good way transfer RF between the backplane and the endcaps (you know, half of the sinks/sources of RF on the satellite). The backplane can't reach the top side of the +Z endcap and the connectors that *would* go to the -Z endcap interfere with the -X frame. Here's a screenshot of the MMCX connectors that *would* connect to the + side of the -Z endcap (legs of the MMCX connectors interfere with the frame): 

[image]

# Possible solutions
We could reach the bottom side of the -Z endcap if we make a cutout in the frame or cut the legs of those MMCX connectors to be hecka stubby. We might be able to remove the MMCX connectors from the card 0 (directly below the +Z endcap) and add some to the bottom of the +Z endcap. (This would amount to a scooch in the +Z direction for the holes for the card 0 MMCX connectors on the backplane, to redirect them to attach to the +Z endcap.)

# Needed action
I'd like an opinion from @glennl and/or @andrewgreenberg. Would it be feasible to modify the MMCX connectors on the backplane (just in the -Z endcap position) so they don't stick out the back? I'm guessing we strongly want to avoid card-to-endcap connections, since we can't slide in a card-endcap pair (endcaps screw, but don't slide).

andrewgreenberg, 2017-10-02:
Since the "end" cards are used to attach the solar panels, it can't just be pulled out like all the cards -
 the side solar panels have to be removed first. I suggest we continue this trend, and make the Z endcaps veritcally plug directly into the "end" cards. So you'll need to remove the two side solar panels AND the Z endcap to remove that top (or bottom) card.

Joedang, 2017-10-06:
Reason for closing:  
We're going with inter-board connections between the head/tail cards and the endcaps, making the head/tail cards big mezzanines between the backplane, solar panels, and endcaps.

closed 2017-10-06 by Joedang

Joedang, 2017-10-06:
Update:  
This is the sort of shenanigan I went with. Note that this is definitely not the final design, since #13 is still a thing.

## tail card (mezzanine to -Z endcap):
[image]