oresat/oresat-structure #13: spacing between head-tail cards and endcaps

Joedang, 2017-10-06:
# Problem
The head and tail cards (position 0 and 19, respectively) are too far apart for the connectors to reach the endcaps without modification.

Here, you can see the 20 pin connectors on the -Z endcap *just barely* connect. They need a 2 mm boost (similar story for the +Z):
[image]

Similarly, the MMCX connectors between the boards (shown below) need a 4 mm boost:
[image]

# Proposed solution
We could have a little "booster seat" board that sits underneath one of the 20 pin connectors, and have a similar thing on both ends of the MMCX connectors. I would imagine you could just solder it to the larger board along with its connector. 
[image]
I'm not sure if that's a sane thing to do electrically or vibrationally. Do any electrical people have thoughts on it? ( @eyand @andrewgreenberg @glennl @aSmig )

aSmig, 2017-10-06:
I don't like it.  This results in solder connections that we can't inspect.  It may not be so important with the 20-pin connector if the booster seat is soldered around castellated edges.  But the RF connector is much more sensitive to pointy solder formations hidden between boards.

What would it take to move those boards closer together?  Why are we considering using board to board connectors with different mated heights?  Are there taller variants of these parts?

andrewgreenberg, 2017-10-06:
I agree with Kenny; a good idea Joe, but not one we should do. What is the actual spacing between boards?

Joedang, 2017-10-06:
> What would it take to move those boards closer together?

Right now the slot, the holes for the connectors, and the connectors are
the starts of arrays. So, the array would need to be moved and individual
features added. It wouldn't be hard to do, but it kind of sucks adding
exceptions...

> Why are we considering using board to board connectors with different
mated heights? Are there taller variants of these parts?

These are just the connectors that are used for everything else. I haven't
looked for taller connectors. If we can use something that works without
moving the boards, that would be ideal.

> What is the actual spacing between boards?

The array is spaced at 10 mm. So 8.4 mm between the boards.

Joedang, 2017-10-06:
Correction:  
The spacing between the head/tail cards and the endcaps is actually 10 mm, not 8.4 mm.

andrewgreenberg, 2017-10-06:
Excellent! OK, then this means that we can get 20 pin connectors that will do exactly 10 mm, and now we just need to find 10 mm RF connectors. Will do!

Joedang, 2018-01-23:
At this point, it seems like the antenna capstone _probably_ won't put electronics on the endcaps. So, I'm going to close this, as it's not really relevant anymore.

closed 2018-01-23 by Joedang