oresat/oresat-structure #26: Shift Alta solar panel "down" 0.5 mm

andrewgreenberg, 2018-02-13:
In the current CAD, the Alta 5 cell module has four slots the tabs go through. The top two slots are underneath the Aluminum frame, so I shifted the entire module down (slots and all) by 0.5 mm.

Please update the CAD and drawings of the solar panels.

andrewgreenberg, 2018-08-05:
Joe: Can you please verify this?

andrewgreenberg, 2018-08-05:
Probably done, but we should really verify this.

andrewgreenberg, 2018-10-07:
OK ignore all the stuff above: The solar connectors are wrong. Change to the through hole versions, and then verify the PCB outlines and BSTOP layers against the new DXF from Jeremy.

jalouke, 2018-10-07:
The drawing is updated to include the center location of the connector. The change is on the "footprints" page.

andrewgreenberg, 2019-04-29:
Done!

closed 2019-04-29 by andrewgreenberg