oresat/oresat-structure #34: First CAD for End Cards and End Caps

andrewgreenberg, 2018-08-05:
This is blocked on extracting the antenna capstone CAD (issue #33). Once that's done, we need four more steps:

1. Take a first pass at the +Z end card with the helical deployer. Probably needs fasteners and other "things", including solar connectors.
2. Take a first pass at the -Z end card with the turnstile deployer. Probably needs fasteners and other "things", including solar connectors. 
3. Take a first pass at the +Z End cap; it has to let the helical antenna deploy without catching. Post drawing so we can make a prototype cap.
4. Take a first pass at the -Z end cap; it has to let the turnstile antenna deploy without catching. Post drawing so we can make a prototype cap.

calvin-young, 2018-10-10:
First-pass CAD for the antennas is in the assembly.  I will continue to update the helical CAD while John (and/or whoever else is assigned) will be in charge of the turnstile.

andrewgreenberg, 2019-04-29:
Old.

closed 2019-04-29 by andrewgreenberg