oresat/oresat-structure #38: Deepen the spring pin holes on the ends of the rails

Joedang, 2018-12-16:
With the presently specified hole depths, the spring pins on the -Z ends of the rails don't sit deep enough. When machined, there was some significant variation in how deep the threads/holes went. So, the threaded case of the spring pin sits out too far. Just deepen the holes and extend that chunk of aluminum to make sure it doesn't break through. (Actually, should we be venting that hole?)

IanZabel, 2018-12-19:
I would personally only recommend venting the hole if it is a press fit or excessively deep (like one order-of-magnitude greater than diameter). This one may not warrant it.

andrewgreenberg, 2019-04-29:
This was just done. Don't need to vent, because the spring pins themselves have a hollow core which allows them to vent.

closed 2019-04-29 by andrewgreenberg